bradcfisher/OctoPrint-ExcludeRegionPlugin #1: Causes the print to print ultra fast

mexusbg, 2018-07-06:
Installing this plug-in caused my printer to not follow the set speed and print at the maximum speed possible, which is not cool at all!

bradcfisher, 2018-07-06:
I'm sorry to hear that it had that affect for your print.  This is a new plugin and has only been tested by myself, so it may have some bugs.

Could you provide some more details to help me track down what the issue may be?  A sample gcode file and copies of the relevant portions of octoprint.log and serial.log would be very helpful.

mexusbg, 2018-07-06:
I will provide gcode, but I have uninstalled the plug-in and made some prints, so for now I cannot provide logs - I don't have that much free time.

The idea for the plug in is great!

[Shoe stopper +3.zip](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2170768/Shoe.stopper.3.zip)

mexusbg, 2018-07-06:
[logs.zip](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2170869/logs.zip)

OK here are the logs. Hope you can find what you need.

bradcfisher, 2018-07-07:
Thank you for providing those files.

I believe I have identified the problem and should have a fix released soon.  The plugin is rewriting retraction recovery commands and setting the feed rate for moves the the retraction feed rate.  If subsequent moves didn't override that feed rate, stuff would print faster than intended.

mexusbg, 2018-07-07:
@bradcfisher I can confirm I don't experience the issue with the last build.
Just a quick question, can I change the feed rate during a print or it may cause the issue again.
I started a very long print and don't want to mess it up.

Thank you!

bradcfisher, 2018-07-07:
@mexusbg Thanks for letting me know the new version is working better for you.  Your feedback has helped improve this plugin quite a bit, and is much appreciated.

Changing the feed rate percentage (M220) or flow rate percentage (M221) should work fine, as the plugin will not modify or prevent those commands.  OctoPrint's feed rate slider on the control tab uses M220, so should be good to adjust through OctoPrint or using the knob/menu on your printer.

Good luck with your print :)

mexusbg, 2018-07-07:
OK I can confirm the issue is gone :)

Thanks!!

brandstaetter, 2018-07-08:
I still have the issue with 0.1.1 it works correctly with deactivated plugin, but if I activate it, I get massive speedups during the layers (outer perimeter at max speed looks great, but on the first layers I like to go slower 😁

bradcfisher, 2018-07-08:
@brandstaetter Please let me know if reinstalling solves the problem.

It is possible you may have a version of the plugin reporting 0.1.1, but still having incorrect feed rate behavior, if it was installed from the master branch instead of the release tag while I was working on the fixes for this issue.  I haven't been using branches when committing new updates, which allows master to get out of sync with the current release.  I'll switch to using a separate development branch for upcoming features/versions, which should help if that's what happened in this case.

If uninstalling/reinstalling the plugin doesn't help resolve the problem, it would be helpful if you could provide a gcode file that exhibits the behavior and a copy of octoprint.log from a print of that file with the plugin enabled.