bradcfisher/OctoPrint-ExcludeRegionPlugin #2: incompatibility with "PSU control" plugin

cmock, 2018-07-06:
After I installed this plugin, the "PSU control" plugin (https://github.com/kantlivelong/OctoPrint-PSUControl version 0.1.7) stopped working; looking at the terminal tab in octoprint, the gcode commands to turn on the power supply do not show up, the user interface parts of the plugin seem to work fine.

After disabling "exclude region", "PSU control" works again (and the relevant commands do show up in the terminal tab).

As I'm using the klipper firmware, PSU control is not sending G- or M-codes, but "SET_PIN PIN=power VALUE=1".

bradcfisher, 2018-07-07:
Thanks for the report.  I have confirmed this behavior, and should have a new release containing a fix soon.

cmock, 2018-08-05:
I'm sorry for the long delay, I only now got around to properly test this, and sadly, it doesn't work.

The relevant lines from the log seem to be these:

    2018-08-05 12:50:46,483 - octoprint.plugins.psucontrol - INFO - Switching PSU On
    2018-08-05 12:50:46,484 - octoprint.util.comm - ERROR - Error while processing hook excluderegion for phase queuing and command SET_PIN PIN=power VALUE=1:
    Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3064, in _process_command_phase
        hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_excluderegion/__init__.py", line 437, in handle_gcode_queuing
        gcode = gcode.upper()
    AttributeError: 'NoneType' object has no attribute 'upper'

This seems to be because of the special klipper commands which are not G- or M-codes, see "Extended G-Code Commands" in https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

Octoprint 1.3.9
ExludeRegion 0.1.1

bradcfisher, 2018-08-05:
Based on the line numbers from the log messages, it looks like your Octoprint instance is still running the code from v0.1.0 and not the v0.1.1 code with the fix.  Could you try uninstalling the plugin completely, and then reinstalling?  I've also had others mention having issues with the plugin being reported as up to date when it really wasn't, and a reinstall was necessary to fix that.

cmock, 2018-08-08:
Yes, but I can only do this next weekend, so please be patient.

cmock, 2018-08-13:
I can now confirm that the upgrade to 0.1.2 has fixed it.

bradcfisher, 2018-08-14:
Thanks for the confirmation.  I'm glad the issue has been resolved, and greatly appreciate your time and effort in helping make the plugin better.

closed 2018-08-14 by bradcfisher