bradcfisher/OctoPrint-ExcludeRegionPlugin #3: Plugin blocks manual extrusion

BerndJM, 2018-07-07:
Manual extrusion from the control tab works only once, than all further extrusion commands are blocked until I do a manual retract, after that again one extrusion is possible.
[octoprint.log](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2172296/octoprint.log)

bradcfisher, 2018-07-07:
Thanks for your report.  I can confirm this is a bug in release 0.1.0.  My initial testing with the upcoming release indicates it is fixed by updates I have made for other issues.  I'll also be making some more changes before that release to disable the gcode filtering when there isn't a print in progress, which will help avoid similar issues like this in the future.

bradcfisher, 2018-07-07:
I've released 0.1.1 which fixes this issue.  Please let me know if you encounter any more problems.

closed 2018-07-07 by bradcfisher

BerndJM, 2018-07-08:
Can confirm that this issue is fixed, thanks a lot for this great plugin ;-)