bradcfisher/OctoPrint-ExcludeRegionPlugin #4: Excluded region persists to next print

andrewebling, 2018-07-12:
I just had to use ExcludeRegion for the first time and it worked perfectly during the print, after the first of 5 separate items came off the bed.

However after the second print item came off the bed, I cancelled the print in Octoprint.

The previously excluded region was then applied to the next print I did (same print as before, but re-sliced with a brim), even though the excluded area didn't show up in the gcode viewer.

I then cancelled the print again, restarted Octoprint and restarted the previous print, which proceeded normally.

Steps to reproduce:

1. Start a print
1. Exclude a region
1. Observe printer stops printing in the excluded region
1. Cancel the print
1. Start a new print
1. Observe the excluded area is not displayed in the Gcode Viewer
1. Observe the excluded area is still applied to the new print (BUG)
1. Restart Octoprint
1. Restart the print
1. Observe the excluded area is NOT applied

bradcfisher, 2018-07-12:
@andrewebling Thanks for reporting the issue.  I'll take a look and see if I can find a way to prevent that from happening.

Currently, the plugin retains the excluded regions for the current file (even after a print is cancelled) and clears them when a new file is selected.  It sounds like replacing the currently selected file with a new version is not detected as a new file selection, and thus didn't clear the regions recorded by the server in that case.  It should have still displayed them in the gcode viewer, so maybe there's a bug there too.

A quick work around would be to select a new file and then back after uploading a new version.  Restarting OctoPrint will also work, but it might be faster to switch files.

bradcfisher, 2019-08-25:
I haven't had any other reports of previously excluded regions being applied, but not showing in the Gcode viewer, nor have I experienced it myself.  Since it may no longer be a problem, I'm closing this issue.  Please feel free to reopen if you continue to experience this behavior.

closed 2019-08-25 by bradcfisher