bradcfisher/OctoPrint-ExcludeRegionPlugin #5: Not an issue: Praise from the community!

foosel, 2018-07-12:
I don't know if you monitor the official OctoPrint forum, so I thought I'd just quickly drop you a link to the praise about this plugin that's happening over there: [Click](https://discourse.octoprint.org/t/exclude-region-plugin-is-one-of-the-best-ever/2742)

👍

closed 2018-07-12 by foosel

bradcfisher, 2018-07-12:
@foosel Thanks for letting me know!  I have not been following the forum and it's great to hear that my plugin has been useful to someone.  Guess I'll have to subscribe :)

BTW: Thank you for all the effort you've put into OctoPrint.  It's a very nice piece of software that makes my 3D printer a much more enjoyable hobby.

foosel, 2018-07-13:
You are welcome, and happy to hear :)