bradcfisher/OctoPrint-ExcludeRegionPlugin #6: ignore command G10 and G11

chinoox, 2018-07-28:
for some reason the G10 and G11 commands for Marlin do not work with this plugging.
Uninstall the plugging and problem solved ....

chinoox, 2018-08-01:
> @chinoox <https://github.com/chinoox> thanks for letting me know. Would
> you be able to provide a gcode file that exhibits the behavior that I could
> use for testing?

This gcode have g10 and g11

bradcfisher, 2018-08-05:
Confirmed as a bug.  v0.1.1 reports the following error:

  File "/home/pi/devel/OctoPrint-ExcludeRegionPlugin/octoprint_excluderegion/__init__.py", line 878, in handle_G10
    returnCommands = self.recordRetraction(RetractionState(firmwareRetract = True, originalCommand = cmd))
TypeError: recordRetraction() takes exactly 3 arguments (2 given)

bradcfisher, 2018-08-06:
I've published a new release v0.1.2 that should resolve this issue.  Please let me know if you still encounter problems with G10/G11 after upgrading.

bradcfisher, 2018-10-02:
Closing, as issue was resolved by release 0.1.2

closed 2018-10-02 by bradcfisher