bradcfisher/OctoPrint-ExcludeRegionPlugin #7: Excluding Regions Where Print Would End Results In Nozzle Being Rammed Into Part

tophattwaffle, 2018-08-09:
I had enough plastic for 1 of 4 parts in a print, so I figured I'd exclude 3 of my 4 parts.

While printing this file and excluding the regions in orange:
[image] This file is printed part by part meaning each part completes before moving onto the next.

With an ending script of:
`;*****ENDING SCRIPT*****
M220 S100 ; Reset feedrate to 100% incase it was changed
M221 S100 ; Reset flowrate to 100% incase it was changed
M104 S0 T0 ; Set Extruder Temp
M140 S0 ; turn off bed

G91 ; Relative XYZ
M83 ; Relative E

G1 Z0.2 F3600 ; Z up quickly
G1 X-10 Y10 E-5 F6000 ; Move XY quickly while retracting. Move backwards to not ruin top finish
G1 Z5 F3600 ; Move Z up some

G90 ; Absolute XYZ
M82 ; Absolute E

G1 X0 Y280 F3600 ; Move bed
M84 ; disable motors
M106 S0 ; turn off fan
;*****ENDING SCRIPT END*****`

The single parts completes and then proceeds to ram the head down into the first part when finishing the print. I'm guessing this means that all gcode movement commands in the regions I selected are canceled. The only thing I can imagine is that it ignores
`G1 Z0.2 F3600 ; Z up quickly
G1 X-10 Y10 E-5 F6000 ; Move XY quickly while retracting. Move backwards to not ruin top finish
G1 Z5 F3600 ; Move Z up some
`
from my ending script because it would have completed these movements inside a canceled region. What I can't figure out is why it attempts moves the nozzle down at all.

The result:
[image]

bradcfisher, 2018-08-10:
Thanks for letting me know about this issue.  I'm not certain exactly why it would have decided to move the nozzle down, but there are a couple of possibilities that come to mind.

If the final move was seen as a move out of an excluded area into a non-excluded area, it _may_ have been attempting to reset the Z position to the position it determined the nozzle should have been at had it printed the part within the excluded region.  I think that's a bit unlikely in this case though, as it appears all of the parts would have the same height after printing.

Another possibility is that I have a bug in how I am dealing with relative moves.  If the plugin was treating the relative move as an absolute move, then it may have been trying to move the nozzle to 0.2mm above the bed, instead of moving it up 0.2mm relative to the current position.

Could you post a simple gcode file that can be used to reproduce the problem?  It may be useful to help diagnose the underlying cause.

tophattwaffle, 2018-08-11:
Once I return from being away this weekend I'll get you a gcode file to
test with.

Lacehim, 2018-11-20:
I had the same issue and came here to report it. :) I've figured out why it happened too.
Took me a few goes to figure it out, and had to rebuilt my entire X axis due to damage to my cooler damaging the bearing housing! :(  Not a major loose though I upgraded it with PETG X axis parts. :)
Anyway, the problem.
When your printing if you are sync'd with job process ticked, and exclude a region it doesn't seem to have a problem, and I was using this perfectly fine on my Prusa in exactly this way.  When I used it with the Wanhao V2.1 it crashed the head into the part and started printing at layer 1.  So I got it back printing, and excluded the region again when a part failed again and it crashed again.  That's when I realised that I didn't have sync to job ticked.  I had been scrolling through the entire job, making sure what I didn't miss a part of the excluded region, and had returned to layer one (scrolled to the bottom on the right scroll bar).
So it seems that when you exclude the region on a specific layer, not the current one, that it resumes printing from that incorrect layer.
I confirmed this, when I excluded the region deliberately to test and selected to exclude a region when sync'd and it excluded the part perfectly fine and continued to finish the remaining parts.

Hope that helps, awesome plugin!  Love it. :)

bradcfisher, 2018-11-21:
@Lacehim, thanks for providing more details about the conditions where this occurs, and I'm sorry to hear that there was damage caused to your printer.

I'm not sure why the behavior would be dependent on the selected layer in the gcode viewer, but I'll definitely give that a try to see if I can reproduce.

If you happen to have a gcode file that you could post, that may also be helpful.  Different slicers structure the gcode differently, and I based most of my testing using gcode from Cura, so it may be a factor if you used a different slicer.

Lacehim, 2018-11-25:
@bradcfisher no worries about the printer, it was a good excuse to do an upgrade :)  I put a new printed X axis on so I can remove X for maintenance without removing any Z components.  Works really well. :)

I was using Cura 3.3.1, and can't for the life of me remember what I was printing to be honest.  It only happened when I wasn't syncing the print in viewer, once I ticked the box and made sure I was in sync I tried again and it was successful but I did that with my finger on the power button.  Hope that helps.

bradcfisher, 2019-08-25:
I've just released version 0.2.0 of the ExcludeRegion plugin (hopefully better late than never).  The new version has a feature that may be helpful to avoid the issue you encountered.  You can now add commands to your begin and end Gcode to ensure they are always executed even if they result in moves in an excluded region.  

For example, by adding

> @ExcludeRegion disable

as the first line of you end gcode, you can ensure the ExcludeRegion plugin doesn't filter any of the gcode commands.

You can do something similar for start gcode scripts, disabling exclusion at the beginning of the script and enabling exclusion at the end.

> @ExcludeRegion disable
> ... other start gcode here ...
> @ExcludeRegion enable