bradcfisher/OctoPrint-ExcludeRegionPlugin #8: Excluded regions do not clear when a print finishes when printing same gcode.

tophattwaffle, 2018-08-09:
If a print finishes with excluded regions, and another printer is immediately started, the excluded regions remain active. Is there a possibility for a tick box to make this a toggle setting? If used for failed parts, and the user want to print another run of those parts it is likely that the user will forget to remove the excluded regions before starting the next print.

bradcfisher, 2018-08-10:
Not clearing the exclusion regions after the print finishes was an intentional design decision I made when implementing the plugin.  I can see how that could be considered undesirable behavior in certain situations.

Based on your suggestion, I'm thinking of doing the following:

1) Add a plugin preference to specify a default action to take when a print finishes (either Retain or Clear the regions)
2) Add a checkbox to the gcode viewer interface that indicates whether to clear or retain the exclusion regions when the print finishes.  When a new file is selected, the checkbox would be populated with the default from the preference.  Toggling the checkbox would change the behavior while the current file is selected.

Would that sufficiently address your concerns?

tophattwaffle, 2018-08-11:
That would be perfect!

bradcfisher, 2019-08-25:
I've released version 0.2.0 of the ExcludeRegion plugin which exposes a setting to control this behavior.  You can now control whether the excluded regions from the previous print are retained or automatically cleared when the print finishes.

closed 2019-08-25 by bradcfisher