bradcfisher/OctoPrint-ExcludeRegionPlugin #9: Printing pauses.

marc9876, 2018-08-12:
Hi I have a problem with the printer pausing midprint momentarily and then resuming. I had a look at the log and this is what i see. The print starts normally enough but this happens mid way through a long print.

2018-08-12 10:20:37,287 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:37,380 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:37,380 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:44,958 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.1573, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:45,126 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:45,282 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:45,283 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:45,614 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.19302, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:45,732 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:45,823 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:45,824 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:47,117 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.09466, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:47,220 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E-0.03600 F1800.00000
2018-08-12 10:20:47,221 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:47,417 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:47,418 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:48,476 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.13429, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:48,542 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E-0.03600 F1800.00000
2018-08-12 10:20:48,543 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:48,579 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:48,580 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:49,474 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.13384000000000001, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:49,601 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.031149999999999997, "feedRate": 1800.0, "originalCommand": "G1 E-0.03600 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:49,639 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:50,103 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.11879, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:50,249 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E-0.03600 F1800.00000
2018-08-12 10:20:50,250 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:50,521 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:50,525 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:50,660 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.09752, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:50,743 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:50,779 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:50,780 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:51,757 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.10351, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:51,860 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E-0.03600 F1800.00000
2018-08-12 10:20:51,861 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:51,897 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:20:51,898 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']
2018-08-12 10:20:52,057 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 0.14584, "feedRate": 1800.0, "originalCommand": "G1 E-0.08000 F1800.00000", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 E-0.08000 F1800.00000']
2018-08-12 10:20:52,122 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E-0.03600 F1800.00000
2018-08-12 10:20:52,124 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:20:52,200 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
[... cut ...]
2018-08-12 10:21:14,889 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E-0.03600 F1800.00000, returnCommands=[u'G1 E-0.03600 F1800.00000']
2018-08-12 10:21:15,032 - octoprint.plugins.excluderegion - WARNING - Encountered recovery without a corresponding retraction: G1 E0.80000 F1800.00000
2018-08-12 10:21:15,035 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 E0.80000 F1800.00000, returnCommands=[u'G1 E0.80000 F1800.00000']

bradcfisher, 2018-08-12:
Thanks for letting me know about the problem.

Could you post a gcode file that has that issue?  It could be very helpful to see the actual sequence of gcode commands that lead to this occurring.  The slicer and settings you used may also be helpful.  

From what I tell from the log snippet you posted, it looks like there's some very small retractions during the print which are not always matched with a corresponding recovery/extrusion.  As long as you don't have any exclude regions defined, it shouldn't affect the print at all, though if that gcode were executed inside an exclude region, it's possible it may cause the filament position to be off after exiting the region to start printing again.

I think the pauses you are experiencing are likely due to the logging itself.  There are several messages being logged per second, and that can slow down Octoprint to the point that the printer is executing commands faster than Octoprint is sending them.  As a workaround, you could try changing the log level for the plugin to ERROR or CRITICAL which may help with the pauses by reducing the amount of detail written to the log.  You can do that through the settings dialog:

[image]

bradcfisher, 2019-08-25:
I've (finally) released 0.2.0 of the ExcludeRegion plugin with updates to address the long pauses when excluding due to excessive logging.  There will still be some delay while skipping the excluded Gcode, but it should be much shorter.

closed 2019-08-25 by bradcfisher