bradcfisher/OctoPrint-ExcludeRegionPlugin #11: Printer timeout when NOT using exclude region actively?

bolsoncerrado, 2018-09-28:
Hi there,

I was wondering if you believe the plugin could be causing timeouts while speaking to my MK2.5 Prusa Printer running 3.4.0 firmware and a Pi3B+ running Octoprint 1.3.9 on raspbian Jesse. Do you think the timeouts near the end of the .log could be related to your plugin even if we were NOT using it during this print? (aka no areas defined at all):

Thanks.
[octoprint.log](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2428351/octoprint.log)

bolsoncerrado, 2018-09-30:
Nah, the temporary pauses are caused by the newest firmware AND the filament sensor being ON. I already reported this to their Github. That error (with the filament sensor) is one that the printer autorecovers fine OR asks you to replace the filament.

The error i'm talking about here is total loss of communication with the printer. 

I indeed can deactivate the plugin, but it happens so randomly that kills my butt, as all our Prusas run the same setup (dedicated Pi, same plugins, same settings, etc.)

Yikes!

Thanks for the input tho.

bradcfisher, 2018-10-02:
I was thinking that those messages from the filament sensor may have been related to the timeout, as they seemed to be related in the log.  From what I was seeing, it looked like at least some of the timeouts were occurring between the firmware generated retraction (G1 E-3 F200) and the corresponding extrusion (G1 E3 F200).  Of course, it's possible the timeouts aren't directly related to the filament sensor issue and they only appear to be so in this case.

Unfortunately, I don't have a Prusa Mk2.5 of my own to experiment with, but I do have a few questions whose answers may help determine whether the issue is related to this plugin:

* How often does the issue happen?
* Have you encountered this issue with the filament sensor turned off?  If it only happens when the sensor is turned on, it may simply mean the printer firmware doesn't respond on the USB port for some reason when the sensor isn't detecting the filament.
* Since you have multiple identical setups, have you encountered the issue on multiple printers?  An issue with only one printer (or a small number) may indicate more of a physical hardware issue, assuming all else is equal.
* Does it occur at the same point when printing the same Gcode file, or does it seem completely random?  If it's the same point in the print, it may be more likely to be the plugin, especially if it doesn't fail when the plugin is disabled.
* Have you tried different USB cables between the the Pi and printer?  Could be due to a loose cable or intermittent connection.
* Have you tried using a different power supply for powering the Pi?  I have heard that using an under-powered or low quality power supply for a Pi can lead to some hard to diagnose issues.

That's about all I can think of for now.  Please let me know if you do happen to find evidence that the exclude region plugin seems to be at the root of the issue.  Any analysis of why that may be the case and/or suggestions for improvements to correct such a problem would be greatly appreciated.

bolsoncerrado, 2018-10-02:
- How often does the issue happen?

Not so often, and so random, I hate these kind of "bugs" hehe

- Have you encountered this issue with the filament sensor turned off? If it only happens when the sensor is turned on, it may simply mean the printer firmware doesn't respond on the USB port for some reason when the sensor isn't detecting the filament.

Currently testing that!

- Since you have multiple identical setups, have you encountered the issue on multiple printers? An issue with only one printer (or a small number) may indicate more of a physical hardware issue, assuming all else is equal.

So far only 1 printer affected.

- Does it occur at the same point when printing the same Gcode file, or does it seem completely random? If it's the same point in the print, it may be more likely to be the plugin, especially if it doesn't fail when the plugin is disabled.

Random!

- Have you tried different USB cables between the the Pi and printer? Could be due to a loose cable or intermittent connection.

I checked the wires already and I do use hi quality ferrite shielded USB cables....

- Have you tried using a different power supply for powering the Pi? I have heard that using an under-powered or low quality power supply for a Pi can lead to some hard to diagnose issues.

Same. I use official 2.5A power supplies. I used to use 3A chinese clones that showed up the yellow lighting bolt on screen and even failed to power properly!

bradcfisher, 2019-01-13:
Closing this report, as it's been a while since it's had any activity.  I hope you were able to successfully resolve the unexpected timeouts you were experiencing.  If you are still having problems and feel they are related to the Exclude Region Plugin, please feel free to reopen and provide additional details.

closed 2019-01-13 by bradcfisher