bradcfisher/OctoPrint-ExcludeRegionPlugin #14: define regions do not clear. must reboot to remove them. gcode not displaying also.

hashtagdavi, 2018-11-12:
i dont know what has changed or why this is going on. plugin was magnificent,  a real game changer for my ability to multi print.

i have changed nothing. 

gcode visualizer is also not displaying any gcode at all. for sliced prints or imported gcode. nothing. 

and exclude region selections are all werid now. cannot properly assign any as they are not functioning correctly when placing. 

and they will not clear out either. have to reboot system for them to go away!

please help !!!

hashtagdavi, 2018-11-12:
[image]

bradcfisher, 2018-11-17:
I'm sorry to hear that you have been experiencing issues with the plugin.  I can't think of any reason that the plugin would stop working if nothing were modified with your Octoprint instance.

Is it possible that your Octoprint instance was updated to a 1.3.10 release candidate version?  I have received a report that the plugin is not compatible with the 1.3.10 release candidates, and has behavior similar to what you describe after upgrading.

bradcfisher, 2018-11-17:
Another possibility may be due to a change in your browser.  I have also had reports of adblockers interfering with editing exclude regions.  If you use an adblocker, you could try turning it off or whitelisting your Octoprint instance.

bradcfisher, 2019-01-13:
Closing this report, since there hasn't been any activity in a while and I have not been able to reproduce any of the issues reported.  There was an incompatibility with Octorprint 1.3.10 and an older plugin version that caused some behavior similar to what is described (see #15 ).  That was fixed by plugin version 0.1.3.

If you are still experiencing any of the issues described, please feel free reopen and provide further details on how to reproduce.

closed 2019-01-13 by bradcfisher

hashtagdavi, 2019-06-25:
HI !!

I sincerely Apologise @bradcfisher  , I did not have notification emails setup for my GitHub account so i had no idea of your response ! 

At the time, yes, I had just updated to the most recent beta RC , which is when the odd behavior started for me. Prior to that I had no idea whatsoever. 

It turned out that it was in fact  AdBlocker that was creating my " issue " 🤦

I don't know why it began blocking only after the update at the time, though of course as you suggested, whitelisting my octoprint instance eliminated the issue entirely !

I have been smoothly printing and eliminating problem parts on my build plate , while completely filling the plate each time I print now. 

You single handedly changed my production capabilities and turn around time to clients , exponentially. More than 10 fold , seriously. 

Thank you so much for your plugin and the time & work you have invested in creating this legitimately ground breaking feature for octoprint. 

I cannot overstate how much I appreciate your work. It has become indispensable to me. 

Thank you so much & thank you as well for your replies. Sorry I didn't get back to you sooner!

Happy printing mate,

David.

bradcfisher, 2019-06-26:
Hi David.  Thanks so much for the kudos and feedback.  I'm glad you were able to resolve the problem you were having, and very happy to hear that the plugin has been beneficial to your productivity.  It's great to get such an overwhelmingly positive endorsement, and confirmation that my work is helping others :)
-Brad