bradcfisher/OctoPrint-ExcludeRegionPlugin #15: Doesn't work properly in  OctoPrint 1.3.10rc2

ChrisHeerschap, 2018-11-15:
Running the latest RC, there were some issues with the gcode viewer in RC1 which are fixed, but when I tried to kill a region that had failed during a print, it wasn't working properly. When I tried to  draw a rectangle, the first click acted as though it were the second with an arbitrary point in the upper corner of the window, and when I tried to resize the rectangle, I started panning the gcode view, with the rectangle remaining fixed.

Excluded the octoprint page from my adblocker to avoid the issue discovered in #13 - but that didn't fix it.

ChrisHeerschap, 2018-11-15:
Absolutely love the plugin, by the way, hope any breaking changes in 1.3.10 aren't a big deal, because I would hate to print in a world that doesn't have this plugin anymore.

bradcfisher, 2018-11-17:
Thanks for the report.  I have confirmed the behavior you describe with the newest release candidate and  I'm currently investigating the cause.  Unfortunately nothing obvious yet, but hopefully I'll have some time to dig into it deeper over the weekend.

I'm glad you've been finding my plugin useful.  It's always great to get such positive feedback 😄

foosel, 2018-11-28:
Shot in the dark, but there were some changes made to the GCODE viewer in foosel/OctoPrint#2717 (that was actually your's I just realized). There were also some changes to the authentication on the socket made necessary by removing read-only mode through the forcelogin plugin, but I don't think they should be relevant here. Otherwise I can't remember doing anything particular to the UI.

ChrisHeerschap, 2018-11-28:
Mentioned this issue in the Octoprint RC3 feedback channel, sounds like the devel version is working.

ChrisHeerschap, 2018-11-28:
By the way, if you like hearing positive feedback, you might want to check: https://discourse.octoprint.org/t/exclude-region-plugin-is-one-of-the-best-ever/2742

bradcfisher, 2018-11-29:
The bug is in how the plugin is detecting and hooking into the updates made to the Gcode viewer and not anything wrong in Octoprint.  I have updated the develop branch with a fix, though that branch also contains other more intrusive updates that I would like to test more thoroughly before releasing.

I'll publish a bugfix-only release to address the issues the plugin has with 1.3.10 and bump the other features to the next release in case I don't find the time to complete that testing.

bradcfisher, 2018-11-29:
Version 0.1.3 has now been released to address this issue.  Please let me know if you experience any additional problems with the plugin in Octoprint 1.3.10.

bradcfisher, 2018-12-01:
No worries.  It was something I should have identified and fixed even before the first RC, especially since I did make the changes in Octoprint that exposed the issue in the plugin.

I haven't received any reports of problems with the new release, so I'm going to close this issue.  Please feel free to open a new issue if any additional problems arise.  Thanks for helping make the plugin better 😄

closed 2018-12-01 by bradcfisher

ChrisHeerschap, 2018-12-03:
Confirmed success in  RC4, thank you!