bradcfisher/OctoPrint-ExcludeRegionPlugin #18: exclude region filters G3 and G4 commands

NatroNx, 2018-12-04:
when exclude region is enabled, G3/G4 is filtered and not sent to the printer.

Marlin supports G3/G4 ARC movements

bradcfisher, 2018-12-04:
Thanks for the report!

The plugin is intended to process the G2 and G3 arc commands similarly to how it processes G0/G1 commands. The slicer I have been using doesn't generate arcs, so the code that does that processing is relatively untested.  Given that, it's possible they are not being handled appropriately.

If you have a sample Gcode file containing some G2 and/or G3 commands that you could attach to the issue, that would be very helpful for analyzing the problem.

NatroNx, 2018-12-05:
Hi,

this reflects my initial findings - you can find a short gcode in it

bradcfisher, 2018-12-16:
Thanks for the example.  I've verified this is a bug in the implementation, and will have a fix in the next version.

calonmerc, 2018-12-28:
Any updates on a release for this? Timeline, etc? Any testing needed? As far as I can tell, neither Slic3r nor Simplify3D use arcs, but in some custom start code I'm testing, I am using an arc, of course!

bradcfisher, 2018-12-28:
@calonmerc, there are some related updates in the devel branch, but I haven't had much time lately to devote to testing them.  I'm hoping I can devote some time to it in the next few days, though I can't give any guarantees.

If you'd like to give the development version a try in the meantime, you can install it using this URL: https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/devel.zip

I believe arcs should work with the development version, though there's a chance there may be some lingering bugs.  If you do happen to give it a try, I'd welcome any feedback you may have about your experiences with it.

calonmerc, 2018-12-28:
First test with the devel branch, it performed the arc perfectly. Running a calibration cube now, but will run a few parts later and exclude one or two of them!

calonmerc, 2018-12-28:
Tested removing a square item, worked as expected. Any other specific testing you'd like on it?

bradcfisher, 2019-01-12:
@calonmerc Thanks for testing out the development version.  I really appreciate the feedback, and am glad to hear the arcs were processed better.

I have some updates in the works that I'll be merging back into the devel branch.  I've been working on implementing more formalized unit tests and apply several other code quality tools.  It's a work in progress, but will reduce the manual testing effort needed for future releases.  I'm working to clean up things for the next release, but, since my time has been very limited lately, I'm not able to give a timeline yet.

bradcfisher, 2019-08-25:
I just released version 0.2.0 of the ExcludeRegion plugin which contains several bugfixes for arc processing.  The new version should not only allow arcs to be printed, but it should also exclude them properly.

closed 2019-08-25 by bradcfisher