bradcfisher/OctoPrint-ExcludeRegionPlugin #19: [QUESTION] Deleting excluded regions

kz2327, 2019-01-01:
Hi!
I was just wondering if it were possible to delete/shrink an excluded region? From what I have gathered, Modify Region only seems to have the ability to expand on any regions that have already been created, and if the original region was sized slightly too big, then it would be impossible to shrink them back down.

bradcfisher, 2019-01-12:
You can currently only modify or delete existing regions when there is not any print in progress.

That was an intentional design decision, since I had originally figured that it wouldn't make much sense to exclude layers under an overhang, and then allow more to be printed where stuff was previously excluded, since anything printed there would be an unsupported overhang and likely to fail.

There are situations where support isn't necessary for higher layers, so I'm now leaning toward relaxing that restriction to allow full modification of the exclusion regions at all times, and putting that determination fully in the hands of the user.

trevorray51, 2019-01-13:
What kz2327 is saying is that once the exclude region has been placed it cannot be removed. The region remains active for all print job to follow. I too have run into this issue. The only way for me to remove the excluded region is to disable the plugin from octoprint, otherwise every print I do excludes the selected regions.

bradcfisher, 2019-01-13:
@trevorray51 The exclude regions do persist following the print, but they should be fully modifiable as long as there is no print in progress and can be expanded during printing.  The regions should also clear automatically whenever a new file is loaded.  When the regions automatically clear will be configurable in an upcoming version, so it can be switched to also clear the excluded regions when the print completes instead of leaving them.

If you are seeing behavior where the regions cannot be deleted or shrunk when nothing is printing, and they do not clear when loading a new file, I believe that sounds like a different issue than described by kz2327.  What you are describing sounds like you may be experiencing something similar to issue #4 , which I have not had any success reproducing.  If you happen to know reliable steps to reproduce, I'd be very interested in that.

Againstyou, 2019-03-19:
> You can currently only modify or delete existing regions when there is not any print in progress.

Can you please add a function to remove excluded Regions while printing?

I had a 10 hour print today and one support break down. The printer from there on just print in the air to build up the support. I looked at the layers and added a "exclude area" to exclude the support. My plan was to try to print the thing without the support. When the support ends and the "real" area start to print, i try to remove the "excluded area" but i wasnt possible. I had to stop and throw away my whole print. If i knew this before i wouldnt have added a "exclude area".

If you can add this feature and make it possible to remove and fully edit the Regions it would be great.
Thanks for your great work so far.

bradcfisher, 2019-08-25:
I've just released version 0.2.0 of the ExcludeRegion plugin which has a setting that can be enabled to allow complete modification of an excluded region at any time.

closed 2019-08-25 by bradcfisher