bradcfisher/OctoPrint-ExcludeRegionPlugin #20: Region modification is limited

Charly333, 2019-01-06:
A small pice of support structure is broken. So i defined an exclude region not to have the filament clashed against the wall of a bearing element in that print. After reaching the final hight of the help structure I wanted to disable that exclude to keep the usable part above beeing printed. But i found it didn´t work I was only able to enlarge the former region I could not move, make it smaller or remove it at all. Please add this possibilities of changes for there is some need in it.

bradcfisher, 2019-01-12:
I have been considering relaxing the restrictions on modifying regions during a print.  I'll try to target that change for the next release.

Againstyou, 2019-03-19:
Had exactly the same issue. Destroyed 10 hour printing time. :(
Thanks for considering to change this and gave us more possibilities.

bradcfisher, 2019-08-25:
I've just released version 0.2.0 of the ExcludeRegion plugin which has a setting that can be enabled to permit modification of excluded regions at any time.

closed 2019-08-25 by bradcfisher