bradcfisher/OctoPrint-ExcludeRegionPlugin #21: excluding doesn't skip retract of skipped region

harakiwi, 2019-01-11:
My gcode prints 3 parts, using Slic3r PE with the _Print Settings_ / _Output options_ / _Complete individual objects_ enabled

In the Gcode viewer i skip the first part BUT i doesn't skip retracts.

So it starts purging, stops at the first objects position, "skips" the first object and keeps retracting the filament.
The second starts but it runs out of filament instantly.

bradcfisher, 2019-01-12:
The plugin should only allow the first retraction in the excluded region to proceed, and all others should be skipped until leaving the region.  At that point it is supposed to recover the skipped retraction and continue on.

If you're seeing excessive retraction, it sounds like that may not be working as intended, which could indicate a bug in the plugin or some properly of the gcode that was being processed that is unexpected.

Would it be possible for you to supply a sample gcode file, and brief description of the area to exclude for reproducing the behavior you're seeing?  That would be very helpful in tracking down the issue.

harakiwi, 2019-01-16:
Tried to answer 2 days ago and something messed / up crashed chrome while i was writing this.

This happened on multiple gcode created especially to print individual parts and me being able to skip a failed part. I created a test file with simple movements (cylinder) to save time testing and reproducing the problem. 
Skipped the print after ~5 layers. worked. skipped the next one. worked.

Trying to find my last projects right now.

harakiwi, 2019-01-21:
Okay so i did a reinstall on a different mSD. Created a fresh 2 object gcode. I let it run for a few layer / minute and excluded the currently printing first object.

Attached the gcode, a short clip showing my filament retraction and the terminal output at this moment.

As i am writing this post it's still "skipping" movement commands but stays above the cancelled object.
UPDATE: while uploading the files it finally skipped the first object and now my filament sensor knows something is wrong and stopped the second object.
[VID_20190121_130908.zip](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2778867/VID_20190121_130908.zip)
[heroforge-testprint-A.zip](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2778870/heroforge-testprint-A.zip)
[terminal-output-A.txt](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/2778871/terminal-output-A.txt)

bradcfisher, 2019-08-31:
Hey, I'm very sorry that I haven't had time to look into this further the past few months.  I appreciate all of the great information you provided, and the video and log definitely show the behavior you described, which is pretty odd.   Only the first retraction should have been excluded, and yet I see the same behavior when I try printing your gcode locally as well, so it definitely appears to be a bug.

I'll take a deeper look into the underlying cause, and can hopefully find a resolution soon.  Hopefully better late than never.

bradcfisher, 2020-06-27:
A (late) diagnosis of the issue is below.  Correction has been committed to the devel branch and will be included in the next release.

This bug is due to the plugin not being compatible with the way Slic3r performs retractions when wiping or retract on layer change are enabled.

Slic3r initally performs a small percentage of the full retraction, in your case 10% or 0.08mm, then it executes the wipe while also retracting the remaining amount (0.72mm) over several XYE moves.  Something similar occurs when retraction on layer change is enabled.

The plugin was implemented using Cura generated Gcode which generates retractions differently (or at least I didn't have any similar options enabled), so this scenario for retractions was not accounted for.

-- Relevant settings from the gcode --

```
; retract_before_travel = 2
; retract_before_wipe = 10%
; retract_layer_change = 1
; retract_length = 0.8
; wipe = 1
```

-- Example of retraction on layer change --

```
G1 E-0.08000 F1800.00000          ; Initial retraction (10%?)
G1 F5760
G1 X162.229 Y101.111 E-0.06152    ; X/Y move with retraction
G1 E-0.65848 F1800.00000          ; Remaining retraction to equal 0.8mm
```

Adding it all together:
-0.08000 + -0.06152 + -0.65848 = -0.8 total retraction

-- Example of retraction with wiping enabled --

```
G1 E-0.08000 F1800.00000        ; 10% retract before wipe
G1 X162.399 Y102.844 E-0.09023  ; several wipe movements, each with a portion of the retraction
G1 X161.969 Y102.616 E-0.14437  
G1 X161.903 Y102.405 E-0.06573
G1 X161.790 Y102.441 E-0.03532
G1 X161.790 Y103.030 E-0.17484
G1 E-0.20951 F1800.00000        ; Remaining retraction following wipe to equal 0.8mm
```

Adding it all together:
-0.08000 + -0.09023 + -0.14437 + -0.06573 + -0.03532 + -0.17484 + -0.20951 = -0.8 total retraction

When the plugin encountered retractions without an intervening recovery, it was simply permitting the
retraction to occur.  The assumption made was that all retractions would always be perfectly paired with
recovery commands (e.g. a single Gcode non-move retraction command would be executed for the full
retraction amount, then a single non-move recovery command would be executed to restore the filament
position).  Since the retraction commands output by Slic3r don't follow that behavior, and include both
non-move retraction commands as well as retractions while moving, the plugin was allowing most of the
retractions to continue.

bradcfisher, 2020-11-17:
A new version 0.3.0 with a correction for this issue and Python 3 support has been released

closed 2020-11-17 by bradcfisher