bradcfisher/OctoPrint-ExcludeRegionPlugin #22: Does not retract while excluding a region

Treez97, 2019-01-30:
I had a problem when using this application. I was printing several pieces when one broke free from the print bed. I went ahead and told it to exclude that part although when it got to that point in the print, the hot end paused above one of the other parts while it skipped that region. During this pause, filament continued to ooze out even though there wasn't being any extruded and ended up in a blob of filament left of the part just before excluded region. Once the next layer reached that same spot, it hit the extra filament and broke that part free from the plate. This continues to happen through all my parts as I would tell it to extrude one and it would then pause over the one before it and leave a bit of filament sticking up. This could easily be resolved by telling it to retract any time it reaches an excluded region and it would be nice to have some setting to adjust it when it comes to an excluded region. Thank you for reading and I hope I can get this fixed.

bradcfisher, 2019-02-01:
Thanks for the feedback.  I have thought about adding configuration options to specify whether a retraction should be automatically inserted when entering an excluded region, but I'm not sure yet how I'd want to implement such a feature.

The plugin already attempts to ensure that retractions already present within the Gcode file are performed appropriately, despite other extruder moves are prevented while excluding.  Because of that, my thinking has been that it shouldn't be necessary in most cases, since a retraction should already occur between separate parts when retractions are enabled in the slicer.

It is possible that a retraction wouldn't happen if only a portion of a single part which doesn't include any retractions is excluded, though the behavior in that case will be somewhat unpredictable anyway.  Other retractions may not be detected if they are combined with moves, so it's also possible those are not currently being accounted for properly.

I'll consider adding settings to force a retraction when entering an excluded region, if no prior retraction was detected.  I suppose that could potentially be useful in some situations.

Another thing to consider, if you are experiencing long pauses while skipping an excluded part, is that there may be some extra logging that you can disable to speed things up and reduce the pauses.  See [these instructions](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/issues/9#issuecomment-412350396) for a way to reduce the logging.  That may be more effective at reducing oozing than additional retraction.

phehr2, 2020-02-19:
I too am experiencing this issue.  As a workaround I tried to add a retract to the "Enter Exclude Region GCode" followed by a feed to the "Exit Exclude Region GCode".  The retracts and feeds did occur, but they happen in such quick succession that the nozzle still leaves a trail of filament behind.  Should this work?

[image]