bradcfisher/OctoPrint-ExcludeRegionPlugin #23: Installed and enabled, but doesn't show up

kazooless, 2019-02-27:
I haven't done any trial and error yet but I've installed the plugin and it shows as enabled. However, I see it nowhere. There isn't an additional tab and there isn't a place to configure it with the rest of the plugins. My first thought is maybe it conflicts with the other plugin I've installed called "cancel objects" but I haven't tried uninstalling that yet.

bradcfisher, 2019-04-22:
Thanks for the report.  I'm very sorry for the delayed response, and I hope you were able to resolve the issue you were encountering.  In case it's still useful:

The Exclude Region plugin simply adds a couple of new controls to allow interactively defining regions within the Gcode viewer in Octoprint.  The current release version of the plugin doesn't have any configuration options, though that will be changing with the next release.  Below is a screenshot of the Gcode viewer with Exclude Region installed.  Note the "Print Exclusion Regions" controls circled in red:

[image]

I haven't used the Cancel Object plugin myself, though it does appear that there have been some updates made within the past few months to include Gcode viewer integration similar to the Exclude Region plugin.  I did try installing them both on the same Octoprint instance, and the controls for both of them did show up, and Exclude Region controls appeared to be functional.

[image]

The homepage for the Cancel Object plugin does mention a potential unintended behavior when used in conjunction with my plugin which I was not aware of, and will investigate whether there's anything I can do from my end.  I'll leave this issue open as a potential bug/compatibility issue while I look into that, though at this point I do not know whether I can do anything about it or not.

kazooless, 2019-04-22:
Thank you so much. I did finally figure it out. Maybe explicitly pointing it out in the read me document would be a good idea, unless it is already there and I just missed it. 

Thanks again for the follow up.

Jeff

bradcfisher, 2019-04-27:
I'm glad you figured out how to use the plugin.  I've been working on the next release which will have updated documentation that should make how to use it much clearer.

I also took a look at the potential incompatibility with the Cancel Object plugin, and it does look like something that can be addressed, but will require changes to both plugins.  The next version of the Exclude Region plugin will have the necessary changes for my end, and I'll try to submit a PR to the Cancel Object plugin which should address the changes needed there.

bradcfisher, 2019-08-25:
I've released version 0.2.0 of the ExcludeRegion plugin which contains updates to improve compatibility with the Cancel Object plugin.  I also submitted a PR to the Cancel Object maintainer with the changes necessary for that plugin.  The new version also has more detailed documentation which hopefully clarifies how to use the various features.

closed 2019-08-25 by bradcfisher