bradcfisher/OctoPrint-ExcludeRegionPlugin #24: Can't Delete Region

Fabrizio3, 2019-03-20:
so, the exclusion works wonderfully, but i finished the print and the area is still there, no print is on at that moment. i click modify, then delete.... area is still there.
try again and same, i go to restart octoprint from my linux pc and see a message that says "cannot delete region while printing", wich is absolutely wrong as no gcode is loaded at that moment
basically could it work if you disabled that safety feature? or at least i could disable it in config?
because the area is stuck there forever as it thinks theres an ongoing print, also if i wanted to change the area mid print it probably wouldnt let me that way
is this a bug? can it be fixed? i really like the functionality and it would be a shame to have to disable the plugin to actually be able to print normally

SuperRoach, 2019-03-25:
I came here to post the second thing.

The use case for this is excluding a region with a hard blob, and wanting to skip a single layer for it, then resume printing there

bradcfisher, 2019-04-22:
@Fabrizio3 Thanks for the report.  The issue with not being able to delete the region after the print finished has been reported in the past as well (see #4 and #19), but I have been unable to reproduce it myself.  The plugin relies on Octoprint server events to notify it of the current printing state, so it may mean an event is not being received by the plugin when the print completes or is cancelled.  I do have plans to relax the restriction on shrinking or removing an exclusion area in an upcoming release, which will allow additional modification during printing. (see #20)

I am also considering adding the ability to define starting and ending layers/z height for which an exclusion region should be active.  That may address the use case mentioned by @SuperRoach of skipping a hard blob, as well as others like skipping supports that failed and then automatically re-enabling printing in that region after the print reaches a height beyond the support.  (See #25)

I believe those changes should address the concerns on this issue, so I'll mark this one as a duplicate.  If there's a nuance I missed, feel free to add some additional details/clarification and reopen.

closed 2019-04-22 by bradcfisher