bradcfisher/OctoPrint-ExcludeRegionPlugin #26: Request: Turn down verbosity of console

paukstelis, 2019-04-25:
This plugin spits out tons of stuff to the console/logs. While this might be useful for debugging, it is mostly unnecessary for the average user. Could you add a debug flag for all the logging?

bradcfisher, 2019-04-25:
Thanks for the report.  This is a known issue, and will be addressed in the next release.  In the meantime, there is a work around for the current version.  Simply change the log level for the plugin to ERROR or CRITICAL to reduce the amount of detail written to the log. You can do that through the settings dialog:

[image]

bradcfisher, 2019-04-25:
In my previous response, I was thinking you were referring to the OctoPrint server side logging.  Now I'm wondering if you might have been referring to the browser's console instead.  Please let me know if it was the browser-side logging that you were referring to, as that would be a separate issue from what had been previously reported and I can record it as a feature request.

The next release will address the high volume of OctoPrint server side logging, and will reduce delays caused by that logging when a portion of a print is being excluded.  That in turn reduces blobbing and improves the print quality.

For browser console logging though, there aren't yet any changes in the works.  There's also not a good workaround to change the amount of that logging with the current version (the workaround I mentioned earlier doesn't affect the browser logging).   There shouldn't be a large amount logged to the browser console, it's not as likely to be visible to a typical user, and shouldn't affect the print quality in any way.

paukstelis, 2019-04-25:
Thanks. I was referencing terminal/log file logging.

bradcfisher, 2019-08-25:
I've released version 0.2.0 of the ExcludeRegionPlugin which exposes a setting to control where the plugin logs (either the Octoprint log, or its own log).

closed 2019-08-25 by bradcfisher