bradcfisher/OctoPrint-ExcludeRegionPlugin #27: Blobs when large portions are excluded.

RedLightning42, 2019-06-15:
I was printing a 10 hour print last night, multiple pieces in one STL. One of the parts broke away, the rest were able to complete using this plugin. However, I tried to print the same gcode and exclude everything but the broken part. It's printing just that area, but occasionally pauses and leaves a blob. I'm assuming it's computing the next lines of gcode and skipping the areas I've excluded, and not sure it can be fixed, but thought I'd let you know.

Pi 3B+ with the latest Octoprint on an Ender 3 Pro.

bradcfisher, 2019-06-15:
That sounds like a known issue with logging causing extra delays when excluding a part of a print.  Try changing the logging level for the plugin as described [in this comment](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/issues/9#issuecomment-412350396).  Hopefully that helps reduce the blobbing.

RedLightning42, 2019-06-15:
I don't have Logging turned on for this plugin, and it doesn't show in the Log menu as per the link. Is it turned on in the background and I need to specifically add it then change the level? I'll play with it another day either way.

bradcfisher, 2019-06-15:
Yeah, the logging is implicitly enabled at the INFO level, and the plugin is generating quite a few log messages at that level.  Adding the entry specifically for the plugin as described in that comment changes it to a higher level and reduces the logging.  It should shorten the pauses by quite a bit, though probably won't make them go away completely.

Not that it would help today, but I have been toying with the idea of adding an "off line" exclusion feature, which may be useful for situations like yours.  The idea would be to load the gcode and define regions when not printing, then save out the result of applying them to a new file.  If done that way, there would be no pauses, since the exclusion would be done separate from printing.