bradcfisher/OctoPrint-ExcludeRegionPlugin #30: Slows down serial communication since Octoprint 1.3.12

cmock, 2019-11-02:
Since upgrading to Octoprint 1.3.12, the plugin slows down serial communication so much that my printer stutters.

This is with no exclusion regions defined.

Turning off the plugin brings my setup back to stutter-free printing.

bradcfisher, 2019-11-02:
@cmock Thanks for the report.  I'm sorry to hear that you're encountering issues with the plugin.  I haven't yet upgraded to 1.3.12 myself, but it looks like this will give me some incentive to do so.  I'll give it a try and see if I can reproduce the issue.

bradcfisher, 2019-11-03:
@cmock I have upgraded to 1.3.12 and ran some test prints with and without the plugin enabled, and have not experienced the issue you described.  It may be helpful if you could you provide some more details about your specific setup.  For example:

* Could you provide a gcode file that exhibits stuttering when the plugin is enabled?  It would be helpful to know what slicer was used, and other factors like print speed and whether the file contains a high number of short moves (I tested using a file sliced in Cura 4.3 with a print speed of 60mm/s and nothing but curves, which generate a lot of small moves that need to be processed by the plugin)
* What version of the plugin do you have installed.  I highly recommend using the newest plugin version (currently 0.2.0), which is what I tested.
* What type of hardware are you running OctoPrint on, and what OS?  I did my testing on a Raspberry Pi 3 B running OctoPi 0.16 upgraded to OctoPrint 1.3.12.
* Do you see unusually high CPU load or load averages when printing?  On my hardware I see CPU for OctoPrint at around 20-35% when the plugin is disabled, and about the same or slightly higher (up to about 40%) with the plugin enabled.
* Do you have any custom logging configuration for the plugin assigned in the OctoPrint settings?  If you do, a level of INFO or higher is recommended.  If you happen to have the plugin configured for DEBUG logging, you may experience stuttering like you described due to a high number of writes to the log.

cmock, 2019-11-04:
I don't have access to the printer in question until next weekend, I'll provide gcode and info about logging configuration then.

The questions that I can answer now:

 * the plugin version is current
 * hardware is a RPi 3B+, on octopi 0.16.0
 * octoprint used 100% of one core when the stuttering occured
 * I've tried setting the CPU frequency to 1.2GHz (governor "performance"), which made no difference
 * printer firmware is klipper
 * there are various other plugins installed, I've narrowed the problem down to ExcludeRegion via binary search, but I haven't tested it with only ExcludeRegion and no other plugins

bradcfisher, 2019-11-07:
Thanks for the feedback.  It sounds like your configuration should perform reasonably similar to mine, especially as it's running on slightly more capable hardware.

Running Klipper is a big difference from my setup where I'm running Marlin.  I don't think there should be much of an issue with that, though it's possible.

Are you using only standard GCode commands, or do you happen to use the extended Klipper commands?  It may be possible that the way the exclude region plugin is parsing the commands is inefficient for those commands.  Not sure if that's the case, but it's one thing that stands out between the two.

The other is that Klipper requires the Pi to do more of the processing of the Gcode that the printer control board would normally do, and would also utilize more RAM on the Pi.  It may be worthwhile to look into things like I/O wait time and swap usage when the plugin is enabled.  I do think it's unlikely it's swapping, but that would cause poor performance.

cmock, 2019-11-07:
I have some extended gcode commands in the start section of my gcode, but the printing phase only consists of standard gcodes (generated by Cura 4.3.0).

Klipper uses only little CPU, and my impression from watching vmstat and top during test prints is that Octoprint often uses up a whole core, while klipper only takes a small fraction of another core. 

But I'll give it a go next weekend, is there anything specific I should test or debug?

cmock, 2019-11-08:
OK, now I'm embarrassed -- I had debug logging on for octoprint and the ExcludeRegion plugin, and setting that to INFO resolved the problem...

Sorry for bothering you and thanks for you help!

closed 2019-11-08 by cmock

bradcfisher, 2019-11-09:
Glad to hear it was a simple fix.  It's not a bother, and it's actually good to know the plugin is being used.    At the least, it gave me some incentive to upgrade my own Octoprint/OctoPi setup to get it a bit more up to date. :)