bradcfisher/OctoPrint-ExcludeRegionPlugin #32: Python 3 compatibility

Idus, 2020-04-01:
I was just asked by OctoPrint (v 1.4.0) today to do a Python 3 test (there is a new app for that) and report an issue to all plugin githubs that were found incompatible with Python 3 (if not already such an issue was stated).
Did the test and got back that the your plugin was Incompatible with Python 3 and could not find any other open issue about it - so here it is. :)

Cheers

bradcfisher, 2020-06-11:
Things have been hectic lately, but I'll try to release a version with Python 3 support soon.  It's the top item on my priority list for this project.

ekr1, 2020-06-11:
Yeah, don't worry. Thanks for the feedback - real_life™ has priority!

nxvash, 2020-08-23:
any update on this?

Hypertoken, 2020-09-16:
Did you upgrade to python 3 before checking all your add-ons compatibility? I did! For this add-on use the Python3 branch until official release (probably has bugs, but works in my setup). 

From terminal run; 
~/oprint/bin/pip install -U "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/feature/python3_compat.zip"

bradcfisher, 2020-11-17:
A new version 0.3.0 with Python 3 support has been released

closed 2020-11-17 by bradcfisher