bradcfisher/OctoPrint-ExcludeRegionPlugin #52: Printing Pauses causing blobs to show

TWIISTED-STUDIOS, 2020-09-22:
hello when you upload a file to the printer and you exclude part of that print when you are printing the pinter pauses on the part it can print doesn't move or nothing and then contuines this causes blobs to appear on the print. i do now know if I'm doing something wrong (some help would be awsome)

CaptainSweatpants, 2020-10-17:
Im now noticing this too. When an area is excluded the nozzle will “park” on top of the last model piece until the time elapses it would take to finish the excluded region. This means the nozzle sits on the piece just cooking it for however long it would’ve taken to finish that excluded area. 

Almost had myself a fire last night because of that. Woke up to burning plastic. This is a new issue suddenly. It never did this before. Previously it would just skip the excluded region and continue printing the rest of the models/pieces with no delay or pause.

TWIISTED-STUDIOS, 2021-01-18:
@CaptainSweatpants I belive this to still be an issue to this day aswell I do not believe that this has been fixed as I ran into this issue again myself recently.

austin4bert, 2021-03-11:
Can confirm this is still an issue. Issue seems to be slow processing of gcode commands.  I filled a plate full of parts to print and excluded all but one, setting them to print all at once. Each layer takes 30 seconds to a minute to exclude all regions. 
[image]

This wasn't always a problem. I remember this plugin seamlessly skipping over excluded regions around 6 months ago.

DatenThielt, 2021-04-15:
Can also confirm this is an issue, I print rather large plates, I excluded one that was failing and throughout the print whenever it would normally go over to the next plate it stayed still for 10 seconds, leaving a blob of filament at several sites on the plate

calvinbui, 2021-08-19:
Mentioned [here](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/issues/27) and [here](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/issues/9#issuecomment-412350396), try reducing the logging level to `ERROR` or `CRITICAL`

ThunderWolf19, 2022-06-16:
This is still happening, any updates on fixing this?

kevman28, 2022-10-26:
same issue here as well.

kevman28, 2022-12-09:
> @kevman28 @ThunderWolf19 @szimszon Would any of you be able to provide more details regarding the issues you are experiencing? For example:
> 
> * Is filament actively being extruded when an excluded region is being processed (is the stepper pushing filament through), or is it just filament dripping from the hot nozzle due to the pause?
> * What hardware are you are running OctoPrint on?  Slow hardware, such as a Pi Zero W may run the software, but will probably be too slow to keep up.  That could cause oozing to be more of an issue as it would take longer to skip over excluded gcode.
> * What slicer was used?  This plugin was developed for gcode from Cura, so it may have issues with output from other slicers if their gcode handles extrusion or retraction differently.
> * Can you provide a small gcode file and directions to reproduce the issue with it?

1. I dont believe it is actually extruding, it just sits in place on top of an existing non-cancelled object, I noticed it does run a command at that spot but does nothing. 
2. I have 4 machines all using a Pi 4b , same behavior
3. Prusaslicer
4. ill have to look back, as I haven't used this in a while, but essentially anything I cancel will result in this behavior.

szimszon, 2022-12-09:
1. For me it is actively extruding the extruder start skipping because the head is not moving at all. It is almost like it want's to extrude all the filament needed for the excluded part.
2. Ender 3 v2 - RPI 4b
3. Prusaslicer
4. Right now I have no time to debug it sorry :( Printing mostly one part at a time...

bradcfisher, 2022-12-10:
@kevman28 @szimszon Thanks for the feedback.

It looks like in both cases, the gcode being printed was produced by Prusaslicer.  I haven't use that slicer myself, but I suspect some of the issues may be related to differences in the gcode produced by that slicer compared to the expectations of the plugin.

In the case of long pauses while excluding potentially causing slow oozing an leading to blobs, I could see that occurring if large amounts of gcode are excluded.  OctoPrint and this plugin need to scan through all of the gcode in the file during printing, and the more gcode that is inspected, the longer that process takes.  The plugin is also pretty inefficient with how it handles arc moves, so it is possible that gcode with lots of arcs could perform more poorly.

To avoid leaving the extruder directly over a printed part too long, I think it may be helpful to change the behavior to perform the initial move into the excluded region to move the extruder away from the last part.  Perhaps any non-extruding moves after entering the excluded region should be executed, to ensure enough distance is gained.  In some cases, that may help avoid damaging the part prior to the region, but it will still probably leave open a possibility of blobs or stringing on the next part due to the hot filament oozing.

If filament continues to be actively extruded when an excluded region is processed, that's a concerning issue.  It likely means that extrusions specified by the gcode being processed within the region is not being processed correctly.  If possible, it would be helpful is someone could post a sample gcode file that causes this behavior for analysis.

HackathonMaster, 2022-12-20:
> @kevman28 @ThunderWolf19 @szimszon Would any of you be able to provide more details regarding the issues you are experiencing? For example:
>
> * Is filament actively being extruded when an excluded [...]

1. filament is NOT actively being extruded (other than the general oozing of filaments because of the heat which is standard behavior)
2. Pi 3B+
3. PrusaSlicer (doubt it has anything to do with gcode produced by prusaslicer as gcode is very standard. The bigger the exclusion area, the longer is takes for the pi to run through all the commands)
4. I currently do not have any gcode but here's a potential temporary solution: move the printer head up a but and inside the excluded region to avoid blobs on the actual print. It will still be a problem as filament will keep oozing out, the longer it takes to do the math. 

Long term solution: find a more efficient way to do the math. When I look at the terminal, it seems like the raspberry pi internally runs the gcode but doesn't send it to the printer. Perhaps a way to avoid executing the code entirely?

kevman28, 2022-12-20:
maybe keeping a small circle/object going in the exclusion zone to keep it purging properly as an optional setting might work for some.

HackathonMaster, 2022-12-20:
How is the math done exactly?

tomha2014, 2023-02-23:
Has anybody solved the problem? I was about to open a bug with Creality, HAHAHA.
I started getting the issue this week
Ender 2 Pro which is only 2 weeks old

Things I've tried All using the same gcode file, not recompiled or anything
- Print via SD card on the printer. Result: clean clear and perfect
- Print via OctoPrint  Result: Pauses and blobs
- Changed out the Raspberry 4 to a Raspberry 3 Result: Pauses and blobs
- Changed out the micro SD card on the PI Result: Pauses and blobs
- Rebuild the OctoPrint Server from scratch, NO PLUGINS Result: Pauses and blobs
- Removed the Camera/Turned of Time Lapse Result: Pauses and blobs
- Tried different filament Inland PETG+ -> Voxel PLA Result: Pauses and blobs
- Tuned the E-Steps Result: Pauses and blobs
- Print uploading the file directly to the upload folder on Octo Print Result: Pauses and blobs
- Print using Cura OctoPrint interface Result: Pauses and blobs
- Generate GCode using Simplify3d vs Cura. Print from SD good, else bad.

Nothing has worked. I'm not a hardware guy but almost looks like a USB can not pass data fast enough and it has to cache.

I have 2 other printers Ender 3 V2 and Ender 3 Pro with OctoPrint and each on there own PI, they do not have problems

Not sure what to try next, 
Ideas anybody? Klipper comes to mind :)

johny-mnemonic, 2023-02-26:
@tomha2014 I am afraid you are at a wrong place with your issue.
What you describe is not related to Exclude region plugin but as you suggested it points to an issue with your printer being able to accept the commands fast enough on the USB link.
My Ender 3 have same issue which usually manifests on rounded objects where standard G-Code creates the circle/curve from hundreds of short straight lines. This overloads the board and printer start to stutter resulting in blobs.
The cure for that is called Arc Welder which replaces these short straight line moves with much smaller amount of arc moves lowering the amount of G-Code commands.

Poporacer, 2024-02-06:
@bradcfisher I am experiencing the same blobbing issues and I see that you requested a file, but didn't see anyone posting one. To help here are my observations and file:

1. Is filament actively being extruded when an excluded region is being processed (is the stepper pushing filament through), or is it just filament dripping from the hot nozzle due to the pause? **Yes, the extruder is pushing filament, but it doesn't look like it is anywhere near the amount for the excluded region.**
2. What hardware are you are running OctoPrint on? Slow hardware, such as a Pi Zero W may run the software, but will probably be too slow to keep up. That could cause oozing to be more of an issue as it would take longer to skip over excluded gcode. **Raspberry Pi 3B**
3. What slicer was used? This plugin was developed for gcode from Cura, so it may have issues with output from other slicers if their gcode handles extrusion or retraction differently. **Orca slicer and I think that might be the problem, I have used this plug-in for quite a while and didn't see this behavior until just recently when I switched to Orca...Might be beneficial to support more than one slicer?**
4. Can you provide a small gcode file and directions to reproduce the issue with it? **Attached**
[Large Bulb BaseFinal_PLA_.28x4.zip](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/files/14186657/Large.Bulb.BaseFinal_PLA_.28x4.zip)