bradcfisher/OctoPrint-ExcludeRegionPlugin #53: Cancelling a print region while the gcode window is zoomable seems impossible - perhaps add a shortcut to do so?

SuperRoach, 2020-10-07:
Moving a print region seems impossible - I would like to move it out of the way, or delete a region. 

I think because the dragging motion is taken by the gcode viewer window being zoomable, that it's not doable. Perhaps add a keyboard shortcut to (c)ancel it?

Meesen0743, 2020-12-25:
Also have the same problem, Quiet annoying when you want to use the tool to remove failed support but not the actual part above it.

bepstein111, 2021-07-11:
Same issue here.....quite annoying.

bradcfisher, 2022-12-09:
The "Modify Region" button at the bottom of the gcode viewer should allow modifying or removing regions.

[image]

After clicking Modify Region, you can click on a region to select it, then drag an resize the region to the new position, then click Apply to make the change effective.  You can also delete the region completely using the Delete button that appears after selecting the region.

This appears to be working as I would expect using the Chrome desktop browser, OctoPrint 1.8.6 and ExcludeRegion 0.3.2

Verustus, 2023-07-09:
I have the same problem I can't move it or make it smaller just bigger. I am using octoprint 1.9.1.