bradcfisher/OctoPrint-ExcludeRegionPlugin #58: Feature Request - Alter Speed in Region

Vantiveman, 2021-03-25:
Hi Brad,
The ability to make some kind of change but only for a specific region during gcode printing creates some fascinating possibilities.
One thing I would very much welcome would be the ability to alter the speed of printing (Feed Rate + Flow Rate) by either percentage or by a fixed defined speed, for a defined area. There are many occasions when I am printing multiple objects but where some of these can be printed a quite high speed (eg 65mmps) but other objects that contain towers, holes or screw threads would be much better quality printed at slow speed (eg, 20 or 25mmps). This plugin could do this, I believe, quite easily and would be a great addition.
For example, define one or more regions, and for these regions select the required speed or percentage reduction.
This would enable multiple items to be printed that each have different ideal printing speeds to give the best quality/speed compromise.
Could you please consider this option?
Many thanks for taking the time to review my request.
Al Wilson

bepstein111, 2021-07-11:
I think this is much better achieved in slicer with modifiers. Cura and PrusaSlicer both can do it quite easily.

bradcfisher, 2022-12-09:
@Vantiveman Thanks for the suggestion.  It is an interesting thought, but I do feel @bepstein111 is correct in suggesting using the built in capabilities of slicers instead.

I could perhaps see a use case where you may only have access to the gcode, and not the original 3D models.  In that case, a plugin with the capability you suggest may be useful, but I would see that as a separate plugin from this one as it serves a different purpose.  I would also suggest that it may be better for such a plugin to pre-process the file prior to printing to produce a new file with the modifications, unlike this plugin which processes the gcode in real time while printing.  That would provide better performance during printing as well as making the results 100% reproducible by simply reprinting the new file.

closed 2022-12-09 by bradcfisher

Vantiveman, 2022-12-09:
Hi Brad,

Many thanks for your feedback. I agree that this suggested functionality may be better suited to a new and separate plugin. 

I wish I had the skills to produce such a plugin.

Kind Regards,

Al