bradcfisher/OctoPrint-ExcludeRegionPlugin #62: Objects excluded outside the drawn zone

tatunts, 2021-07-27:
Hello!

This was the first time I've used this plugin, not sure why this happened: two objects had failed mid print (about 50%), so I drew the exclude zone. After that, four more perfectly printed objects got excluded as well.

Here's a video showing:

https://streamable.com/uzor0m

Do you think this a rare bug or can I trust it on another job?

Please let me know if you need any more information..

Thank you.

bradcfisher, 2022-12-09:
Sorry for the very delayed response.

Based on your video, I appears that the four objects that disappear simply do not go above that layer and thus there is no gcode to render for them.  Note how the objects within the excluded region are still visible, despite being within the region.  The region does not prevent the gcode viewer from rendering the gcode, it simply shows overlays indicating where the gcode will be prevented from being sent to the printer.

closed 2022-12-09 by bradcfisher

ASchneiderBR, 2022-12-09:
Hello! No problem. It has been a while now, I don't remember exactly what happened, but I can assure you all parts were exactly the same, those ones that desapears were exactly like the other ones. Thanks.