bradcfisher/OctoPrint-ExcludeRegionPlugin #65: Install on Windows 10 Plugin not loading

mrcsdl, 2022-03-24:
Hey, i tryed to install your Plugin on my Windows 10 Instance. using this Guide.
https://community.octoprint.org/t/setting-up-octoprint-on-windows/383

Everything is working so far, but i have the Problem when i try to install your Plugin ill get an error in the CMD and the Plugin will not show up. 

Here is a log of Octoprint Starting: https://pastebin.com/tcfxP49b
(2022-03-24 15:55:50,313 - octoprint.plugin.core - ERROR - Error loading plugin excluderegion)

Can you maybe help me with that?

Iam Using 

OctoPrint 1.7.3
Python 3.10.3

electricAGENT, 2022-04-27:
Hello, I am having the same issue, but with Windows 11. It will go through the install process, but once it shows the "Plugin Management" blurb in the top right corner it shows as an "Unknown" plugin... I have the latest python and Octoprint as it is shown in the other comment from mrcsdl. Any help would be great, and this plugin is fantastic on every other machine I have running on a pi and has been a real life saver!

jneilliii, 2022-06-03:
I think you might need to install this patched version. In plugin manager copy/paste this URL in ...fromURL at the bottom.
```
https://github.com/jaketri/OctoPrint-ExcludeRegionPlugin/archive/refs/heads/master.zip
```
There's an open pull request that fixes this exact error, just waiting on @bradcfisher to merge and release a new version.

closed 2022-10-31 by mrcsdl