bradcfisher/OctoPrint-ExcludeRegionPlugin #71: can't update

kevman28, 2022-12-08:
error trying to update

/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip
ERROR: HTTP error 404 while getting https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip
ERROR: Could not install requirement https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip for URL https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip

7s3, 2022-12-08:
Same

`Updating, please wait.
++++++++++++++++++++++++++++++++++++
Now updating Exclude Region to 0.3.2
++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip --no-cache-dir
ERROR: HTTP error 404 while getting https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip
ERROR: Could not install requirement https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip for URL https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
`

PaintOnBrush, 2022-12-08:
https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin 

is the link it's searching for, however the repository is actually:

    https://github.com/katiekloss/OctoPrint-ExcludeRegion

dojohnso, 2022-12-08:
It's looking for https://github.com/katiekloss/OctoPrint-ExcludeRegionPlugin/archive/0.3.2.zip, which is this repository we're commenting on (although the other exists as well), however the file it's trying to grab can be found linked here https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/releases/tag/0.3.2, and it is actually https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/**refs/tags**/0.3.2.zip so the update is looking in the wrong spot. 

Will wait for the developer to advise.

bradcfisher, 2022-12-09:
Apologies for the issues.  There was some confusion with the plugin which caused version 0.3.1 to be released from a different location.  Anyone who had updated to that version may run into problems installing the new version 0.3.2.

All that should be needed to resolve the problem is to uninstall the plugin and reinstall a clean copy.